JanGoe/esp8266-wifi-relay #6: Wieviele "freie" GPIO sind noch vorhanden?

NemoN, 2015-12-11:
- Könnte man noch einen DS18B20 anschließen?
- Welcher ist der "RESET" Pin? (für den Anschluss vom Reed-Kontakt)

JanGoe, 2015-12-11:
Ja es ist möglich dht22 oder ds18b20 anzuschließen dafür könnte man den heraus geführten gpio 0 nutzen 

Der reset Pin ist nicht extra heraus geführt hier für müsste man am esp8266 direkt ein Kabel anlöten 
Reset ist der Pin oben links

NemoN, 2015-12-11:
Kollidiert das nicht mit dem Firmware-Modus? GPIO0 -> GND und Restart?

JanGoe, 2015-12-11:
Wenn man dort nur einen Temperatur Sensor anschliest sollte es keine Probleme geben

JanGoe, 2015-12-15:
Hallo,

"Freie" Gpios sind
Gpio4,5,14,15,16 (tested) und 9,10 ( ungetestet)

JanGoe, 2016-01-16:
in der nächsten pcb version werden "alle" freien gpios heraus geführt auf eins, und es wird auch ein 3,3v pin und ein 5v pin geben

closed 2016-01-16 by JanGoe