JanGoe/esp8266-wifi-relay #16: Reconnect bei Router Restart

JanGoe, 2015-12-19:
Bug wenn Router z.b. Neu startet verbindet sich der esp8266 nicht mehr automatisch mit dem WLAN erst nach neu Start

NemoN, 2015-12-19:
Hier hatte jemand ein ähnliches Problem. Er prüft wohl aber ob der noch mit dem MQTT Broker noch verbunden ist...

https://openhardwarecoza.wordpress.com/2015/07/09/updated-mqttesp8266-sketch-to-automatically-reconnect-wifi-and-mqtt-if-the-wifi-went-down/

Ich denke mit einer Art Watchdog, der zum Beispiel alle 5 Minuten prüft:

- ob der ESP8266 noch verbunden ist = eine  IP hat (evtl. auch den Router pingen kann)
- ob noch genug HEAP Speicher frei ist

und dann gegebenenfalls den ESP8266 neu startet sollte helfen.

JanGoe, 2016-01-16:
nach einigen tests kann ich sagen, das der esp8266 sich bei mir eigentlich regelmäßig wieder dem dem router verbunden hat

getestet wurde folgendes:
ESP8266 "online"
router wurde vom Strom genommen, ca. 60sek gewartet, router wieder eingesteckt nach ca. 60sek konnte ich den esp unter der gleichen ip wieder reichen

closed 2016-01-16 by JanGoe

JanGoe, 2016-01-16:
zum testen habe ich mir alle 15sek den heap an meinen webserver senden lassen 

dazu in der init.lua unter dofile("aktor.lua") ein dofile("heap.lua") eingefügut

heap.lua:
function send_to_visu()

  print(link)
  
  conn = net.createConnection(net.TCP, 0)
  conn:on("receive", function(conn, payload)
    print(payload) 
  end)
  -- senden an shc
  link = "/debug.php?chipid=" ..node.chipid().. "&heap=" ..node.heap()

  conn:connect(80,"192.168.0.54")
  conn:send("GET "..link.." HTTP/1.1\r\n")
  conn:send("Host: 192.168.0.54\r\n")
  conn:send("Accept: */*\r\n")
  conn:send("User-Agent: Mozilla/4.0 (compatible; esp8266 Lua; Windows NT 5.1)\r\n")
  conn:send("\r\n")
  conn:on("sent", function(conn)
    conn:close()
  end)
  conn:on("disconnection", function(conn)
    print("Got disconnection...")
  end)

end

tmr.alarm(2, 15000, 1, function()
send_to_visu()
end)