JanGoe/esp8266-wifi-relay #24: Statische IP

xCite1986, 2016-01-27:
um eine Statische IP einzubinden, hab ich in die 5. Zeile in der init.lua ergänzt

[image]

JanGoe, 2016-01-27:
Hallo,

Danke dafür allerdings würde dies einen kleinen fehler verurschen das es dazu kommen könnte das der TCP server nicht startet

die prüfung ob die ip nicht null ist wird "übersprungen" und somit startet direkt die aktor.lua und der tcp server und das obwohl wlan evtl noch nicht richtig verbunden ist

ich lasse mir hierfür eine lösung einfallen

xCite1986, 2016-01-27:
keine Ursache.. oh. okay.. also sollten hierfür zwingend die WLAN-Einstellungen stimmen.

mein Schrottmodem kann nämlich keine IP´s an MAC-Adressen zuweisen.

Vielleicht macht die Integration einer Config-html nach der erstmaligen WLAN-Konfiguration sinn um diverse Variablen zu setzen (Verriegelung, low/high invertiert, Wlan-Settings, DHCP/StaticIP, DHT ein/aus... usw)

JanGoe, 2016-01-29:
das mit der config.html nehme ich mal auf und schaue was ich machen kann

kellermaaan, 2016-04-05:
Hab mal ne Frage, ist es möglich diesen Befehl: `php tcp.php 192.168.0.62 2x4x1` statt der IP mit dem Hostname zu nutzen? Wenn nicht wie sieht es denn mit der Einstellung der statischen IP aus? Wäre sehr hilfreich wenn man das vielleicht einstellen könnte...

xCite1986, 2016-04-05:
Ich hab das ganze mit der ESPEasy und MQTT gelöst, reconnect funktioniert super, ip's laufen dennoch auf DHCP und das Modul erhält dann quasi einen Namen zum schalten  wie /switch_az/gpio/2 und zur statusübermittlung /switch_az/Switch_1/Switch

Im pimatic-forum ist alles dokumentiert

https://forum.pimatic.org/topic/1257/wifi-switch-to-integrate-in-wall/42#

LG alex

kellermaaan, 2016-04-05:
Ich habe es mal getestet. Wenn man statt der IP den Hostname in den Befehl einfügt müsste es auch funktionieren. Hat zumindest bei mir so geklappt!

xCite1986, 2016-04-05:
Bei mir hat der reconnect des moduls ins wlan nach einem routerreset leider nicht funktioniert...

kellermaaan, 2016-04-16:
Hallo nochmal, ich wollte nochmal nachfragen, ob es nicht eine andere Möglichkeit gibt eine statische IP oder MQTT etc. einzustellen??? Das wäre nämlich echt super!

JanGoe, 2016-04-16:
Hallo,

Welche probleme hast du denn das board zu steuern ?

was @xCite1986  schreibt in sachen MQTT über ESPEays ist sichher eine gute methode, hier kann man auch alles bequem über den Browser einstellen und braucht in lua dateien keine änderungen mehr vor zu nehmen

mit folgenden befehl : wifi.sta.setip({ip="192.168.0.126",netmask="255.255.255.0",gateway="192.168.0.1"})

kannst du dem esp aber auch eine feste ip geben

xCite1986, 2016-04-16:
@kellermaaan 

Ich denke dass aktuell Softwareseitig ESPEasy am flexibelsten und angenehmsten zu bedienen ist.

in der config kannst du via Webinterface statische IP`s vergeben, MQTT konfigurieren, deine angeschlossenen Bauteile an den restlichen GPIO`s verwalten, interne rules anlegen etc...

http://www.youscreen.de/fptulhwcd24.jpg
http://www.youscreen.de/gfqiuvtyg40.jpg

Wie Jan oben schon erwähnt hat, funktioniert nach Verbindungsabriss kein reconnect.
Wenn du willst kann ich dir morgen step-by-step Support im Pimatic-Chat anbieten, 

LG Alex

kellermaaan, 2016-04-18:
Erstmal danke für eure Hilfe aber momentan funktioniert es ganz gut. Aktuell spreche ich das Board über den Hostnamen an und reboote das Board nachts immer so gegen 2 Uhr. Nach einer gewissen Zeit ist das Board immer hängen geblieben... Das passiert nach dem rebooten bis jetzt nicht mehr.
Das mit "wifi.sta.setip({ip="192.168.0.126",netmask="255.255.255.0",gateway="192.168.0.1"})" habe ich auch schon versucht aber damit ist der TCP Server, wie in einem vorherigen Beitrag schon erwähnt, nicht gestartet.

xCite1986, 2018-01-15:
gelöst, kommuniziere nun via mqtt mit dem board... aber nachwievor sind die boards im einsatz.. gute Arbeit.. respekt

closed 2018-01-15 by xCite1986