JanGoe/esp8266-wifi-relay #25: Pimatic-Integration

xCite1986, 2016-01-27:
Hallo Jan.

Erstmal danke für die saubere Arbeit... eine echt geniale "Erfindung" hast du da :) wenn das Relais zuverlässig schaltet, den Zustand an meine HomeControl verlässlich weiter gibt, werde ich noch 5 weitere benötigen... ^^

Ich hoffe du kannst mir bei folgende Anliegen weiterhelfen.

1. 

Bei dem Anschluss an 230V~ zieht das Relais sofort an. die Softwareseitigen ein/aus-Stellungen stimmen zwar, aber wäre es möglich das Relais zu invertieren? also bei Leerlauf soll es nicht anziehen und trotzdem aus sein.

ESPlorer-Debug bei "Ein":

2x4x1
low
2x4x1
low

ESPlorer-Debug bei "Aus":

2x4x0
high
2x4x0
high

 2.

Des weiteren versuche gerade das Relais in Pimatic einzubinden.

Einen Thread gibt es ja bisher im zugehörigen Forum
[http://forum.pimatic.org/topic/1257/wifi-switch-to-integrate-in-wall/](http://forum.pimatic.org/topic/1257/wifi-switch-to-integrate-in-wall/)

In meiner Pimatic-config hab ich folgendes erfasst:

[image]

Bei einschalten, gefolgt von ausschalten schaltet das Relais zwar wunderbar durch, ich erhalte jedoch folgende Errormeldung:

[image]

3. 

Zusätzlich hab ich gegen GPIO12 und GND einen Schalter angeschlossen.
Schalte ich diesen ein und wieder aus erhalte ich folgende Debug-Meldungen im ESPlorer:

/CMD?Switch1=ON
Got disconnection...
/CMD?Switch1=OFF
Got disconnection...

Auf der Pimatic-Übersichtsseite ändert sich jedoch leider kein Schalterstatus :(

Danke und LG 
Alex

JanGoe, 2016-01-29:
Hallo, 

zu 1. das wird sich mit der neuen pcb version ändern der gpio für das Relais wird geändert
2. das ist keine wirklich Fehler Meldung nur eine "notice" 
könnte man einfach fixen mit zeile13 der tcp.php -> @$filename = $argv[3];

3. um das richtig einzustellen muss du die Aktor.lua entsprechend anpassen hierzu kannst du dir die pimatic doku anschauen -> https://pimatic.org/api/actions/ 
da ich zurzeit leider keine pimatic installation habe kann ich hier nicht wirklich weiter helfen kann dir aber anbieten dich via teamviwier zu unterstützen dann können wir uns das mal gemeinsam anschauen

xCite1986, 2016-01-29:
Hallo Jan,

1.) Ich das durchtrennen der Leiterbahnen und verbinden der anderen Relais-Pins behebt das Problem.

[image]

2.) Diese Meldung hab ich weg bekommen indem ich in der tcp.php gelöscht habe

[image]

3.) Das Problem mit der Schalterstatusaktualisierung scheint teilweise behoben zu sein. Im Intervall von 5 Sekunden holt sich Pimatic den Status des Schalters ab - das funktioniert soweit.

Sollte jedoch das wlan ausfallen und der schalter nicht erreichbar sein, fährt die Statusabfrage ins leere und dies würde mir mir die errorlog in pimatic anfüllen :(

Ich hätte gerne dass das ESP-Modul die Schalterstellung in Pimatic verändert, sollte sie ausgelöst werden.

 Die ersten Zeilen der aktor.lua hab ich momentan wiefolgt abgeändert:

[image]

In der ESPlorer-Debug wird folgendes ausgegeben:

[image]

Die URL wäre korrekt und würde den Schalterzustand verändern. Nur denke ich dass diese nicht ankommt (Zustand ändert sich erst bei "abfrage:%")

Danke für das Angebot der Teamviewerunterstützung, melde mich diesbezüglich die kommenden Tage per Mail bei dir.

LG Alex

JanGoe, 2016-01-30:
1. prima werde drüber nach denken nächste pcb versionen gleich so herstellen zu lassen
2. zeile 14 sollte bestehen bleiben für die update funktion der tcp.php
3.

versuch mal zeilen 28-45 zu löschen und mit folgenden zu ersetzten
`link = login.. ":" ..password.. "/debug.php?chipid=" ..node.chipid().. "&heap=" ..node.heap().. "&version=290116e"
conn=net.createConnection(net.TCP, 0) 
conn:on("connection",function(conn, payload)
            conn:send("GET "..link.. " ".. 
                      "Host: 192.168.0.200\r\n"..
                      "Accept: */*\r\n"..
                      "User-Agent: Mozilla/4.0 (compatible; esp8266 Lua;)"..
                      "\r\n\r\n") 
            end)           
conn:on("receive", function(conn, payload)
    print('\nRetrieved in '..((tmr.now()-t)/1000)..' milliseconds.')
    print(payload)
    conn:close()
    end) 
t = tmr.now()    
conn:connect(80,'192.168.0.200') `

welche nodemcu version ist auf deinen chip ?

xCite1986, 2016-01-31:
[image]

was soll diese Zeile bewirken??

login.. ":" ..password.. "/debug.php?chipid=" ..node.chipid().. "&heap=" ..node.heap().. "&version=290116e

[image]

teste deinen code am abend...

JanGoe, 2016-01-31:
link solltest du entpsrechend anpassen :smile:

xCite1986, 2016-01-31:
hallo jan.

hab nun folgendes in der aktor.lua stehen:

[image]

bei tasterdruck passiert aber leider vom ESP aus immer noch kein update der schalterzustandes in pimatic :(

es sollte einfach ein POST von folgendem schema abgesendet werden

```
http://admin:xyz@192.168.0.200/api/device/Licht_Arbeitszimmer/changeStateTo?state=true
http://admin:xyz@192.168.0.200/api/device/Licht_Arbeitszimmer/changeStateTo?state=false
```

xCite1986, 2016-02-01:
vielleicht helfen diese Webseiten weiter.

https://github.com/pimatic/pimatic/issues/347
https://github.com/pimatic/pimatic/issues/60
https://github.com/georg90/esp8266-pimatic-1/blob/master/main.lua

xCite1986, 2016-02-01:
hmm.. ich glaub ich befind mich in der Zielgerade ^^ ... aber iwas passt da noch nicht :dash: 

[image]

nugged68, 2016-02-01:
Ich habe es jetzt mal so hinbekommen....

>   PimaticServer = "192.168.178.31"
>   BaseLoginPimatic ="VXNlcjpQYXNzd29yZA=="
>   -----------------------------------------------
> 
> function send_to_visu(sid, cmd)
>   platform = "Pimatic"
>   
>   if (platform == "Pimatic") then
>     if (cmd == 1) then switch="true" elseif (cmd == 0) then switch="false" end
>     port = 80
>     link = "/api/device/"..sid.."/changeStateTo?state="..switch..""
>   end
>   
>   if (platform == "Openhab") then
>     if (cmd == 1) then switch="ON" elseif (cmd == 0) then switch="OFF" end
>     port = 8080
>     link = "/CMD?" ..sid.."=" ..switch
>   end
>   
>   --print(link) 
>   
>     conn=net.createConnection(net.TCP, 0) 
>     conn:on("receive", function(conn, payload) print(payload) end )
>     conn:send("GET "..link.." HTTP/1.0\r\n")
>     conn:send("Authorization: Basic "..BaseLoginPimatic.."\r\n")
>     conn:send("Host: "..PimaticServer.."\r\n")
>     conn:send("Content-Type:application/json\r\n")
>     conn:send("Connection: close\r\n")
>     conn:send("Accept: */*\r\n\r\n")  
>     conn:on("receive", function(conn, payload)
>     print('\nRetrieved in '..((tmr.now()-t)/1000)..' milliseconds.')
>     --print(payload)
>     conn:close()
>     end) 
>     t = tmr.now()
> 
>     conn:connect(80,'192.168.178.31')
> 
> end

Die Änderung bzw die Umschaltzeit in Pimatic ist so ca. 1 Sekunde.

Hoffe es Hilft dir weiter, um über die Linie zu kommen  ;-)

JanGoe, 2016-02-01:
evtl sollte man noch hinzufügen das du den "Authorization" ich glaube base64 verschlüsselt hast ?
user:password  als base64 oder ?

sonst hilft es glaube ich nicht vielen

nugged68, 2016-02-01:
oh ja, sorry...hab ich vergessen.... 
auf der Seite https://www.base64decode.org/ kann man es encoden

JanGoe, 2016-02-01:
wäre schön wenn ihr evtl einen text für die readme.md schreibt damit es andere user später einfacher haben?

nugged68, 2016-02-01:
Ich werde die Tage bzw.  am Wochenende mal was zusammenschreiben und dir dann per Mail schicken.
- Einstellungen in Pimatic (config.json)
- Anpassungen in der aktor.lua

xCite1986, 2016-02-01:
meine ursprünglich codezeile 

```
conn:send("Authorization: Basic "..login..":"..password.."\r\n")
```

war auch base64 verschlüsselt... dachte das spar ich mir da es auch unverschlüsselt gehn könnte.. sorry, haha... javascript is nicht so meins :dancers: 

ich teste das ganze dann morgen nachmittag und fass mal ein kurzes tutorial zusammen.. 
wenn es dir nichts ausmacht, jan, publiziere ich es auch im pimatic-forum

xCite1986, 2016-02-01:
danke dir übringens @nugged68 für deine unterstützung :D dank dir ist jetzt jan um 6 module ärmer *g*

JanGoe, 2016-02-01:
gerne drafst du es im pimatic forum publizieren 

um 6 module ärmer bin ich doch gerne, aber von der "jetztigen" version sind auch nicht mehr viele da, neue version in ausreichender stückzahl gibt es wohl ende feb.

Die neue version hat dann auch eine Sicherung mit drauf

und weiter freie gpios werden auf pins raus geführt ( 1x 5V pin  mit bei RX,TX,GND)
3,3v,Rest,GPIO16,GPIO5,GPIO4,GND ( neu auf pins )

xCite1986, 2016-02-02:
Hab dir einen Pull bezüglich der README.md gesendet

Werde dann das ganze übersetzen und anschließend im Forum posten.

Na gut.. dann warte ich noch mit meiner Bestellung ;-)

puuh... so viele GPIOs, ich hoff du hast genug Platz auf der Platine... 
ein weiterer Vorschlag wäre den zweiten Relais-Last-Anschluss ebenso raus zu führen.. wenn Platztechnisch irgendwie möglich. Dann würde sich das Relais "Out of the Box" in eine Wechselschaltung integrieren lassen

JanGoe, 2016-02-02:
habe deine readme.md noch etwas geändert und genommen

wenn du 1fache boards willst könnte ich dir schon die neue pcb version zu
nen guten preis anbieten
auch 2fache könnte ich dir mit der neuen pcb version schon anbieten,
allerdings muss ich hier nen kabel anlöten das relay 2 schaltet

danke für die anleitung
mfg
jan

closed 2016-02-03 by JanGoe