JanGoe/esp8266-wifi-relay #26: Zufällige Resets wenn 220V an den Relais anliegt

NemoN, 2016-01-29:
Wenn ich an beiden Relais 220V Verbraucher angeschlossen habe, stürzt das Modul reproduzierbar fast jede Minute mit folgender Fehlermeldung ab:
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

Testaufbau sind zwei 2x LED Lampen die direkt neben dem Modul liegen. 
Evtl. Störstrahlung von den LEDs die den ESP8266 zum Absturz bringt?

Mit Niedervolt Verbrauchern läuft das gleiche Modul mit der gleichen Software Version tagelang stabil...

JanGoe, 2016-01-29:
hast du an den ausgängen die Varsitoren angelötet ?

NemoN, 2016-01-29:
Modul ist voll bestückt: https://www.dropbox.com/s/o0lmibg9osjyx3d/20160129_110107.jpg?dl=0

JanGoe, 2016-02-22:
sollte mit der neuen pcb version und den etwas dickere leiterbahnend für die 250v relevanten teile nicht mehr auftreten

closed 2016-02-22 by JanGoe