JanGoe/esp8266-wifi-relay #34: Ein @ im WiFi Password ergibt keine Verbindung

Gunstick, 2016-03-14:
Mein WiFi Netz hat ein @ im Passwort. Aus unerklärlichen Gründen funktioniert das nicht. Ohne @ ist kein Problem. Leider darf ich das Passwort vom Hackerspace nicht ändern :-)

JanGoe, 2016-03-15:
bist du sicher das es am @ zeichen liegt ?

könntest du mir ein bild posten vom esplorer direkt nach eingebe der wlan daten mit @ deine ssid und das pw kannst du gerne ausgrauen

evtl liegt es an den "50" versuchen alle 150ms es kann sein das dir 50 versuche nicht reichen wegen z.b. schlechtem wlan empfang
um das auszuschliesen bitte mal die init.lua ( https://github.com/JanGoe/esp8266-wifi-relay/blob/master/lua-tcp/init.lua)  änderen zeile 12 'if (count == 150)' then zum testen

Gunstick, 2016-03-22:
Was ich gemacht habe ist ein hotspot auf meinem smartphone eingerichtet. Passwort war helloworld. Funktioniert. Password auf hello@world gesetzt. Geht nicht mehr. Das smartphone lag neben dem relais :-)

JanGoe, 2016-03-24:
Hallo,

Problem habe ich eben gefunden auf deinen board wird nicht die neuste version der servernode.lua drauf sein, mit der aktuellen  -> https://github.com/JanGoe/esp8266-wifi-relay/blob/master/lua-tcp/servernode.lua version

gibt es 2 funktionen hex_to_char und unescape ( zeile 65-71 ) damit werden die probleme mit dem @ behoben

closed 2016-03-24 by JanGoe

Gunstick, 2016-03-27:
installed recent servernode.lua, solved the problem.