JanGoe/esp8266-wifi-relay #35: Rollladensteuerung

kellermaaan, 2016-03-24:
Hallo,

hat jemand mit dem Board schon mal seine Rolläden gesteuert? Ich musste nämlich leider im Internet lesen, dass die verbauten Relais nicht dafür ausgelegt sind...

Wenn das mit den Rollläden funktioniert, hätte ich vielleicht auch noch eine Idee! Wäre vielleicht gut wenn man in einer Konfig oder direkt im Code ne Zeit eingeben könnte wie lange ein Relay sich ein und dann wieder ausschaltet. Dann könnte man sagen "fahr Rollade X für 45ms hoch/runter".

JanGoe, 2016-03-24:
Hallo 

Also ich selbst betreibe mit dem gleichen Relais selbst 14 Rollläden diese laufen Zeit Ca. 1 Jahr ohne größere Probleme das war auch ein Grund warum ich diese verwendet habe 

Sollt er du noch die v1 ( blaue Platine ) haben kann ich dir nur empfehlen Kabel Brücken für die 10a Erweiterung anzulöten Beschreibung dazu gibt es in der readme.md 

Das mit der Zeit sollte sich im Code ohne Problem lösen lassen aber meinst du wirklich 45ms ? Oder euere 45sek ? 
Ich habe das mit der Zeit zurzeit mit einer rule in openhab umgesetzt

Mit freundlichen Grüßen
Jan

kellermaaan, 2016-03-24:
Alles klar, dann weiß ich bescheid. Kann es ja mal probieren!

Also ich habe die v2. Da muss keine Brücken löten oder?

Ach ja stimmt, 45 sek wären wahrscheinlich sinnvoller. Openhab hab ich leider nicht, ich nutze Pimatic. Dort weiß ich aber leider nicht, ob ich in den Rules das ebenfalls umsetzen kann. Eine Lösung im Code würde ich vielleicht gut finden!

xCite1986, 2016-03-24:
die zeit in pimatic zu legen und mit zu übergeben is zwar möglich, aber dadurch dass 1-2 sekunden delay hier eine wertvolle rolle spielen eher eine heikle sache.. reagiert der ESP nicht sofort, krachts übern fenster *g*

jetzt kommt es auch darauf an was "der code" können soll.. nur komplett auf und zu oder soll man auch zwischenstops einlegen können?

aber jan hat dafür bestimmt ein gutes lua als ansatz *gg*

LG Alex

kellermaaan, 2016-03-24:
Ich würde es bevorzugen, wenn man auch Zwischenstopps einlegen könnte. Wobei zu beginn komplett auf und zu erstmal reichen sollte. Code wäre dabei wirklich die bessere Lösung...

Gunstick, 2016-03-24:
Wenn der chip direkt endstopschalter abfragen kann, wäre das auch noch ein zusatz zum lua code. Damit würde der rolladen dann oben/unten schön brav stoppen.

JanGoe, 2016-03-24:
eigentlich haben doch fast alle rolladenmotoren einen eingebauten entabschalt punkt fest im motor drin oder nicht ?

so ist es bei meinen zu mindestens einen für oben und einen für unten

mi-hol, 2016-12-05:
@JanGoe  nach dem merge koennte man das issue doch schliessen, oder?

closed 2016-12-06 by JanGoe