JanGoe/esp8266-wifi-relay #45: http 400 beim openhab update

tkoeberl, 2016-06-28:
Hallo,
ich habe das ganze mit openhab versucht zu konifigurieren. Hat sich bei mir aber etwas quergestellt, daher habe ich direkt per rest API auf openhab zugegriffen:
`function updateOpenhab(host, port, relay_sid,state)
    print("Updating openhab")
	Tstart  = tmr.now()

	conn=net.createConnection(net.TCP, 0) 
	-- show the retrieved web page
    conn:on("receive", function(conn, payload)
	end )
	-- when connected, request page (send parameters to a script)
	conn:on("connection", function(conn, payload) 
	   post_length=string.len(state)
	   print('\nConnected') 
	   conn:send("PUT /rest/items/"..relay_sid.."/state HTTP/1.1\r\n"
		.."HOST: "..host..":"..port.."\r\n"
		.."content-length: "..post_length.."\r\n\r\n"
		..""..state.."\"")
	   end) 	   
	-- when disconnected, let it be known
	conn:on("disconnection", function(conn, payload) print('\nDisconnected') end)
	conn:connect(port,host) 
end
updateOpenhab("openhab_server", 8080, "RollUp", "ON")
updateOpenhab("openhab_server", 8080, "RollUp", "OFF")`

vielleicht kanns ja jemand brauchen..

Grüße,
-thomas

JanGoe, 2016-07-14:
Könnte mir vorstellen das es daran liegt das du für dein obenhab user/pw brauchst ?

tkoeberl, 2016-07-14:
nein, leider ist es das nicht, aber per REST geht es problemlos.

closed 2016-12-06 by JanGoe