JanGoe/esp8266-wifi-relay #46: Boards funktionieren leider nicht stabil!?

kellermaaan, 2016-07-02:
Schönen guten Tag,

ich habe leider ein Problem mit meinen Relais-Boards. Hat es einer geschafft, dass die Boards stabil funktionieren? 

Ein, zwei Tage funktionieren die Boards bei mir wunderbar und sie fahren auch die Rollladen hoch und runter aber irgendwann sind die Boards nicht mehr erreichbar… Ich bekomme dann immer den folgenden Fehler:

-> error [pimatic-shell-execute]: [ShellActionHandler] stderr output from command sudo php /home/pi/tcp.php 192.168.2.111 2x5x1: PHP Warning:  fsockopen(): unable to connect to 192.168.2.111:9274 (Connection timed out) in /home/pi/tcp.php on line 75

Manchmal hilft da einfach ein Befehl zum Neustarten (0x0) und wenn das nicht hilft muss ich die Sicherung rausnehmen und wieder reinmachen damit das Board neugestartet wird. In einer Netzwerk-App mit der man die Netzwerkgeräte erkennen kann, werden die Boards dann auch als offline angezeigt… Momentan sende ich alle 7 Stunden den Befehl zum Neustarten aber das hat es auch nicht wirklich gebessert. 

Hätten jemand vielleicht eine Idee was man da machen könnte??? Das wäre nämlich echt gut, denn das ständige Neustarten nervt ein bisschen. 

Mit freundlichen Grüßen
Kilian Kellermann

JanGoe, 2016-07-02:
Hallo

Was helfen sollte wäre die neusten Lua Dateien auf den esp zu laden 
Oder ggf zum testen jede Minute einen Ping senden oder mit der debug.lua den esp jede Minute z.b. An Pi Matic einen Status zu senden 

Mit freundlichen Grüßen
Jan

kellermaaan, 2016-07-02:
Erstmal danke für die Hilfe!
Das mit dem Ping habe ich schon mal probiert aber das hat auch nicht so wirklich geholfen... Probiere es aber nochmal. Der Vorschlag mit der debug.lua wäre vielleicht noch etwas. Aber wie funktioniert das denn? Ich kenne mich leider mit diesen lua Dateien noch nicht so aus...

prikril, 2016-07-02:
@kellermaaan
Du kannst ja mal versuchen dir eine Firmware auf http://nodemcu-build.com/ bauen zu lassen.
Dauert ein paar Minuten. Dann brauchst du nur noch Python 2.x installieren und das esptool.
Vielleicht wurde ein paar Bugs gefixt, die bei dir Probleme verursachen.
Aber wenn du es dir nicht zutraust, lass es lieber.
Und wenn die alte Firmware bei @JanGoe problemlos läuft, wüsste ich nicht, warum das bei dir nicht auch so sein sollte.
Wie viele wifi-relays hast du? Haben alle das Problem? Wenn ja, sind alle gleichzeitig nicht mehr erreichbar? Sind diese alle V1 oder V2?
Vielleicht hast du ja WLAN Probelme?
Welche(n) Router/Accesspoint(s) setzt du bei dir ein?
Hast du diese(n) mal neugestartet?
Gibt es da Logeinträge, die helfen könnten?

Grüße
Dominik

kellermaaan, 2016-07-05:
Schon mal danke für deine Hilfe!
Also momentan habe ich 3  Boards im Betrieb. 2 steuern Rollladen und ein weiteres eine Lampe. Das Boards welches die Lampe steuert funktioniert einwandfrei, nur die die an den Rollladen angeschlossen sind haben manchmal diese Probleme. Das passiert auch immer unterschiedlich und zeitunabhängig. 
Alle meine Boards sind bereits V2 und müssten glaube ich auch fast die neuste Firmware draufhaben. 
Über ein Wlan Problem habe ich auch schon nachgedacht aber wüsste nicht warum es daran liegen soll. Als Accesspoints haben wir drei dLAN® 500 AV Wireless+. Die da nicht vielleicht die Möglichkeit, dass man mit den Boards die Wlan-Stärker auslesen könnte???

Ich hatte auch jetzt am Wochenende mal die Lösungen mit dem Pingen eingestellt. Bis heute morgen hat es auch wunderbar geklappt nur heute morgen nicht mehr. Was ich letztens auch vergessen habe zu beschreiben ist, teilweise schalten die Boards auch nur vielleicht gute 2-3 Sekunden durch bis sie sich wieder abschalten obwohl ich 45 Sekunden eingestellt habe. Deshalb bewegen sich die Rollladen nur 3-4 cm hoch oder runter. Dieser Fall ist heute morgen nämlich auftreten. Hätte da jemand vielleicht eine Idee was das sein könnte???

Gruß 
Kilian

prikril, 2016-07-05:
Also das sind ja interessante Erkenntnisse.
Wenn das Lampenboard keine Probleme macht, kann es daran liegen, dass dort das WLAN besser ist. Oder wird ein anderes WLAN Netz verwendet?
Mit diesen ganzen Powerlinegeräten, egal welcher Marke, hab ich persönlich noch keine guten Erfahrungen gemacht. Hast du die neueste Frimware auf deinen Adaptern?
Es kann auch sein, dass ein Bug im Rolladen Code ist. Das müsste man sich mal anschauen.
Die WLAN Stärke kannst du ab NodeMCU 1.5.1 mit "wifi.sta.getrssi()" auslesen. Lies dazu https://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifistagetrssi
Die Werte sind z. B. "-87", "-82", "-71", wobei "-82" besser ist als "-71".
Kannst du programmieren?
Ich habe ein Projekt für LDT (Eclipse) gestartet um beim Programmieren Autovervollständigung zu erhalten.
https://github.com/prikril/ldt-nodemcu
Verwendest du bereits den neuesten Code dieses Projekts, also unter anderem:
https://github.com/JanGoe/esp8266-wifi-relay/blob/master/lua-tcp/actuator.lua
?
Wenn du Updatehinweise braucht, melde dich nochmal.
Ich habe nämlich vor einigen eine große Verbesserung der Codequalität vorgenommen.
Das sollte jetzt auch für Programmieranfänger relativ gut zu lesen sein.
Kannst du die Board einfach an den Computer anschließen oder sind diese fest, z. B. in der Wand, verbaut?
Es gäbe die Möglichkeit die Funktion "start_tcp_server()" so zu erweitern, dass du die WLAN Stärke aus der Ferne abfragen kannst.
Das mit dem nur ein paar Zentimeter bewegen klingt allerdings nicht nach WLAN Problemen, sondern eher danach, als würde der ESP neu starten.
Mit dem Befehl "tmr.time()" https://nodemcu.readthedocs.io/en/master/en/modules/tmr/#tmrtime
kannst du abfragen, seit wie vielen Sekunden der ESP bereits läuft.
Wenn er wirklich neu gestartet hat, sollte der Wert recht klein sein.
Auch diesen Wert kannst du aus der Ferne abfragen.

kellermaaan, 2016-07-08:
Ne, es sind alle im gleichen Netz. Also mittlerweile würde ich Wlan-Probleme ausschließen, weil ich momentan nach dem Pingen keine Probleme habe. Das einzige ist, dass die halt zwischendurch nicht komplett hoch oder runter fahren. Bei einem neuen Board, welches ich an ein neues Rollo angeschlossen habe, durfte ich was ganz komisches beobachten... Immer wenn ich runterfahren wollte, ist das Rollo immer nur gute 3-4cm runtergefahren. Egal was ich gemacht habe, selbst bei einem weiteren Versuch oder über die angeschlossenen Schalter ist das Rollo nicht weiter runtergefahren. Ich konnte nur diese paar cm wieder hochfahren. An dem Rollo kann es denke ich mal nicht liegen, weil es ohne Board ohne Probleme hoch und runter fährt. 
Also programmieren kann ich, nur leider nicht dieses NodeJS... 
Ja die Boards sind in der Wand verbaut, deswegen ist es schwer diese an den PC zu schließen...

prikril, 2016-07-09:
Mit welchen SmartHome Programm steuerst du die Wifi-Relays? (pimatic?)
Kannst du mal versuchen die Befehle zum Öffnen und Schließen manuell mit der tcp.php Datei zu senden?
Und kannst du dein SmartHome Programm währeddessen deaktivieren, damit es nicht dazwischenfunkt?
Bzw. deaktiviere dein SmartHome Programm oder deine WLAN Verbindung und versuche das Rollo ausschließlich mit den Schaltern zu steuern.
Geht das Rollo dann ganz auf und zu?
Falls du weiterhin Probleme hast, würde ich dir, um einen Softwarefehler auszuschließen, empfehlen, die aktuellen Dateien auf den ESP zu spielen.
Zusätzlich kannst du auch noch eine aktuelle Firmware aufspielen.
Dazu muss du das Board an den Computer anschließen.
Um die einzelnen Dateien upzudaten kannst du die tcp.php verwenden.
Oder du nutzt den Code, von dem sich anscheinend auch Jan inspirieren hat lassen:
https://github.com/breagan/ESP8266_WiFi_File_Manager
(Allerdings finde ich die API echt schlecht. Die Idee dahinter ist gut.
Denke, dass ich das Projekt in den nächsten Wochen mal forke und diesem Projekt hier anpasse.)
Hinweis: Du brauchst eigentlich nur die PHP Dateien. Dort musst du die Ports von 80 auf 9274 ändern.
(Ich weiß nicht wie gut du programmieren kannst und wie fit du mit der Linux Shell bist.)
Dann gehst du alle Dateien (servernode.lua, actuator.lua, set.html) durch, die ein Update bekommen haben.
Zum Schluss lädst du die neue init.lua auf den ESP.
Dann sendest du den Befehl zum Neustarten.
Wenn dann alles gutgegangen ist, kannst du testen, ob der Fehler weg ist.
Sollte etwas schief gegangen sein, musst du das Board aus der Wand ausbauen und die Dateien oder die Firmware neu aufspielen.
Ich hoffe du hast ein entsprechendes Kabel.

kellermaaan, 2016-07-12:
Erstmal nochmal danke für diese vorbildliche Hilfe! 
Als SmartHome Programm benutze ich pimatic. Also wenn ich das Board nur über die Schalter betätige, fährt das Rollo trotzdem nur diese paar cm nach unten. Das ist richtig komisch... Als ich das Board ganz am Anfang angschlossen hatte funktionierte es auch 2-3 mal und dann auf einmal dieses komische Verhalten...
Momentan habe ich auch nicht die neuste Softwareversion, deshalb würde ich die mal erneuern. Wie funktioniert das denn mit der tcp.php? Könntest du mir das ein wenig erklären? Ich habe zwar ein entsprechendes Kabel aber die verbauten Boards wollte ich erstmal nicht ausbauen.

prikril, 2016-07-12:
Also du hast dein pimatic bzw. dein WLAN deaktiviert, wenn du die Schalter betätigst?
Ich vermute nämlich, dass dir pimatic reinpfuscht. Deshalb bitte erst WLAN ausschalten und sicherstellen, dass der ESP keine Verbindung mehr zum Netzwerk hat. Dann erst die Schalter betätigen.
Und dann sagst du, ob es immer noch nicht geht.
Warum solltest du die Dateien updaten, wenn es bei einem anderen Rollo auch geht?
Was passiert bei deaktiviertem WLAN, wenn du einen Schalter mehrfach hintereinander drückst?
Fährt das Rollo dann immer ein paar Sekunden oder gar nicht mehr und nur noch in die andere Richtung?
Sollten die obigen Schritte nicht helfen, kann ich dir gern erklären, wie du Dateien mit der tcp.php auf den ESP überträgst. Mit der Shell auf deinem Raspberry kannst du umgehen?
Schön ist es zwar nicht, aber es geht und man macht es ja eigentlich nicht dauernd, wenn es mal funktioniert.

NemoN, 2016-07-12:
Ich hatte auch schon mal einen ähnlichen Effekt. Beim Schalten von 220V Verbrauchern wurde zufällig der ESP zurück gesetzt. Nach dem Reboot Cycle war das System wieder erreichbar...

prikril, 2016-07-12:
Kann es sein, dass das verbaute Netzteil nicht gut mit Spannungsschwankungen zurecht kommt?
Aber dann müsste der ESP doch gleich neustarten und nicht erst nach 2-3 Sekunden oder?
Vielleicht sollte @kellermaaan mal das Board ausbauen und Lampen statt des Motors dranhängen, um zu sehen ob es dann geht?
Welche technischen Vorschläge gibt es, um bei einer zukänftigen Version 3 des Wifi-Relays dieses Problem zu lösen?

JanGoe, 2016-07-12:
Hallo,

Also eine Technische Lösung für version 3 wäre einen optokoppler zu
verbauen um das relays komplett galvanisch vom esp zu trennen
Vorschläge wie man das umsetzten könnte sine willkommen

interessant zu wissen wäre auch ob das Relay weiterhin "aktiv" bleibt wenn
der Rollladen gestoppt ist ?
wenn das board neustartet bekommt der motor ja kurz Strom auf beiden Seiten
einige Motoren verstellen dann ihre endabschalt punkte das solltest du ggf
mal prüfen ?

Ich würde vorschlagen mal zum testen ein board auszubauen und wieder den
normalen schaltet einbauen und "testen" ob der Rollladen dann wieder normal
geht, wenn ja dann wirklich mal zum testen Lampen an das board anschließen

Mit freundlichen Grüßen
Jan

:

prikril, 2016-07-12:
Naja das Relais trennt doch den 5V Stromkreis des ESPs vom Stromkreis des Motors. Das ist doch durch die Bauart eines Relais so gegeben. Wichtig bei der V3 wäre nur, dass die Relais nicht dauernd angezogen sind, so wie ich das mit dir bereits besprochen habe. Dann gibt es auch das Problem des "Stroms auf beiden Seiten" nicht mehr.
Das Problem hier ist ja eher, dass das Netzteil des Board und der Motor am gleichen "Kabel" hängen.
Zu vergleichen mit dem Flackern einer Glühbirne, wenn man im gleichen Raum einen Staubsauger ansteckt und einschaltet.
Da würde mich echt interessieren, wenn man mit einem Oszi die Ausgangsspannung des Netzteils aufzeichnet, wie das Netzteil auf Spannungsschwankungen im Netz reagiert.
@kellermaaan hat ja jetzt erstmal einiges zum Ausprobieren...

JanGoe, 2016-07-12:
Wegen den Spannungs-Schwangungen ist ja extra ein 470uf Kondensator vor dem
ESP mit anboard der das unterbinden sollte

Alle aktuellen Versionen die es nun zu kaufen gibt nutzen die anderen
Kontakte vom Relais damit sollte das problem das beide motor seinen Strom
haben gehoben werden

:

prikril, 2016-07-15:
Wenn jemand die Dateien auf dem ESP updaten will und nicht die tcp.php verwenden will oder es sich nicht zutraut, kann sich hier mein Projekt mal anschauen. (ESP8266 NodeMCU WiFi File Manager)
https://github.com/prikril/ESP8266_WiFi_File_Manager
Es ist ein Fork von einem anderen Projekt, das um die Funktion erweitert wurde, dass der Port abgeändert werden kann. (Der TCP Server der WiFi Relays läuft standardmäßig auf 9274)
Die PHP Dateien müssen auf einen lokalen PHP Server kopiert werden.
Dann ist es möglich Lua Dateien, welche in einen bestimmten Ordner auf dem Server kopiert wurden, auf den ESP zu übertragen. (Anleitung siehe Link)
Mein Projekt ist mit diesem hier kompatibel, da es die selbe API wie die tcp.php benutzt.
Was nicht funktioniert, ist die "Status" Funktion, weil dieses Projekt hier die entsprechende Schnittstelle nicht anbietet. Man sieht also auch nicht, welche Dateien auf dem ESP liegen.
Das behindert aber die anderen Funktionen nicht.
Fragen / Anregungen / Fehlermeldungen bitte in meinem Projekt und nicht hier. Danke.
Um Nachrichten über Änderungen zu meinem Projekt per Email zu erhalten, reicht ein Klick auf "Watch".
Zum Schluss noch der Hinweis:
Sollte bei der Übertragung etwas schief gehen, kann es sein, dass der ESP nur noch per TTL Kabel erreichbar ist oder sogar die Firmware neu aufgespielt werden muss.

mnmn2, 2016-10-28:
Hi,
I have stability issue also.
The board reboots quite often,
and the relays are changing state at the restart moment...
Quite strange, as I have a garage-door-opener connected to it...
Any ideas?
thanks,

famthiem, 2016-11-19:
I have had this problem, too.
Since I put the Reset-Pin with a pull-up resistor to VCC, the problem seems to be fixed.

NemoN, 2016-11-19:
Which value has the resistor?

prikril, 2016-11-19:
@famthiem 
How did you find out the solution could be the reset pin?

You could also pull-up or pull-down the transistors for the relays to prevent changing state during startup.

I have the NO version of the board. (relay normally open)
For these you have to change the RELAY_STATE values in actuator.lua
For NO you have to use a pull-down resistor to GND.
I tried it with resistors values between 100k and 150k to GND.
120k and 130k worked best for me.
Maybe you need other values for VCC.
Default version of the board is NC (normally closed relay, opened through VCC)
For those you need a pull-up resistor.

Pull-down to GND works only for relay 2.
Relay 1 is connected to GPIO 2 which produces strange levels during startup.
Quote: When choosing GPIO pins to use, it’s best to avoid GPIO 0, 2 and 15 unless absolutely necessary.
(https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup)

I already have discussed the NO NC thing with the inventor.
He sent me a NO version of the V2 board. But must board are NC.
In my opinion the relays should be NO instead of NC.
The NC relays are opener relays (German: Öffner).
So they consume power if they are switched to "off" state.
And they are closed if the board has no power.
I think they should be open if the board has no power.

The inventor @JanGoe claims that "Alle aktuellen Versionen die es nun zu kaufen gibt nutzen die anderen Kontakte vom Relais" (all newer boards use the other pins of the relay).
But many people already have the NC version now...

kellermaaan, 2016-11-20:
Also ich habe mal versucht die lua-Dateien mit dem (ESP8266 NodeMCU WiFi File Manager) zu aktualisieren. Wie kann ich denn erkennen, dass nun die aktuelle Version auf den Boards laufen? Ich habe alle Dateien in die 'filebin' kopiert und dann auf der Webseite alle hochgeladen. Man muss doch nur auf die Dateien klicken, die im weißen Feld stehen damit die aktualisiert werden, oder???

prikril, 2016-11-21:
Ja die Bedienung funktioniert so, wie du es beschrieben hast.

Leider ist der File Manager nur teilweise mit diesem Projekt hier kompatibel.
Die "Status" Funktionen, also das Auslesen der Systemvariablen und auflisten der Dateien ist aktuell (noch) nicht möglich. Wenn ich irgendwann mal dazu kommen sollte, baue ich das noch ein.
Dann gibts eine neue Version dieses Projekts und des WiFi File Managers.
Bis dahin arbeitet man sozusagen blind und kann nicht direkt überprüfen ob alles geklappt hat.
(Ich habe eine Idee für ein kleines Skript, das man über den WiFi Manager auf den ESP hochlädt und ausführt. Damit könnte man Systeminformationen auslesen und es müsste dieses Projekt hier nicht verändert werden. Allerdings weiß ich nicht, wann ich Zeit habe das umzusetzen.)

Es gibt aktuell nur die Möglichkeit die Version der actuator.lua Datei zu überprüfen.
Mit dem Befehl `php tcp.php 192.168.x.x 9x0` (IP anpassen) solltest du die Version 0.4.0 erhalten.

kellermaaan, 2016-11-21:
Alles klar, danke. Leider hat es nur bei einem von drei Boards richtig funktioniert... 
Bei den beiden, bei denen es nicht wirklich funktioniert hat habe ich immer diese Meldung bekommen:
"Warning:  fsockopen(): unable to connect to 192.168.x.xxx:9274 (Operation timed out) in ~/send_File.php on line 58"
Bei dem anderen Board habe ich das nicht bekommen. Das zeigt mir mit dem Befehl "php tcp.php 192.168.x.x 9x0" auf 0.4.0 an. Allerdings sind alle Boards erreichbar. Die Software Pimatic habe ich zu der Zeit gestoppt, damit nichts dazwischenfunkt.
Hätte jemand vielleicht eine Idee woran es liegen könnte???

NemoN, 2016-11-29:
@prikril 

> For NO you have to use a pull-down resistor to GND.
I tried it with resistors values between 100k and 150k to GND.
120k and 130k worked best for me.

Was bewirken diese Widerstände? Mehr Stabilität?

prikril, 2016-11-29:
Also bei mir ist es so, dass die Relais anziehen, wenn der ESP startet.
Das ist, weil die GPIO standardmäßig auf Input stehen.
Mit dem Pull-Down des Transistors wird das Relais offen gehalten.
Dazu:
http://www.mikrocontroller.net/articles/Relais_mit_Logik_ansteuern
Zitat: "Wichtig ist hier R1. Dieser Pull-Down-Widerstand sorgt dafür,
dass der MOSFET sicher sperrt wenn der steuernde Mikrocontroller sich im Reset befindet oder gerade programmiert wird."
Erst wenn der GPIO auf Output High steht, zieht das Relais an.
Leider geht das nur beim zweiten Relais, da das erste an den GPIO 2 angeschlossen ist, der beim Starten sehr undefinierbare Zustände durchläuft.

NemoN, 2016-11-30:
> Leider geht das nur beim zweiten Relais, da das erste an den GPIO 2 angeschlossen ist, der beim Starten sehr undefinierbare Zustände durchläuft.

Könnte man hier eine Schaltung einsetzen, die den Ausgang (GPIO2) erst nach sagen wir 500ms aktiv schaltet?

xCite1986, 2016-11-30:
Hallo Thomas,

ein Vorschlag wäre die Leiterbahn zu gpio2 abzutrennen und die Verbindung mit einem anderen gpio herzustellen. ich seh mir später an ob das möglich wäre... 

allerdings müsste man das auch im Code dann ändern...

lg alex

NemoN, 2016-11-30:
Hallo Alex, Code Änderung ist kein Problem da ich einen eigenen Homie Sketch verwende. GPIO 2 würde ich als Input nehmen (sofern möglich - da hängt ja auch die Status LED dran) und statt dessen Zb GPIO 4 als Output für das Relais.

NemoN, 2016-11-30:
Homie esp8266: https://github.com/marvinroger/homie-esp8266

prikril, 2016-11-30:
Also man kann den Programmcode schon ändern.
Man müsste die Zeile:
`RELAY1_PIN = 4 -- GPIO2`
in der init.lua und actuator.lua ändern.
Leider ändert sich dadurch auch die TCP Schnittstelle, da beim Switch Commando der Pin mitgegeben wird.
Man kann natürlich auch diese ändern, aber schön ist das nicht.
Die Schnittstelle sollte eh komplett geändert werden.
Das mit der Pin Änderung und Codeanpassung ist dann eher was für ein Board V3.
Ansonsten würde man die Kompatibilität des Codes mit den V2 Board nur schwer aufrecht erhalten können.
Für diese Diskussion wäre eigentlich ein eigenes Issue oder ein Forum notwendig.
Aber vielleicht sollte man die ganze Sache hier nicht größer machen als sie ist.
Man weiß ja nicht wieviele Board schon verkauft wurden, oder?
Neben @NemoN gibts bestimmt auch einige andere die ihren eigenen Code verwenden.
Ich finde es gut, dass es dieses Board gibt.
Den Code sehe ich persönlich eher als eine Art Vorschlag einer Grundkonfiguration.
Diesen kann jeder so anpassen, wie es einem am besten passt.
Allzugroße Sachen laufen auf dem ESP eh nicht.
Wenn man es hinbekommen könnte, dass die Boards stabil laufen mit einer einfachen Schnittstelle (ähnlich wie jetzt), glaube ich wäre das für den Großteil der Nutzer zufriedenstellend.

stefan73, 2016-12-04:
Die Stabilitätsprobleme bzw Resets sind ziemlich sicher die bekannten Reset Probleme, wenn man Wechselstromlasten mit einem klassischen Relais schaltet. Hintergrund: Die meisten Lasten haben heutzutage einen kapazitiven oder induktiven Anteil. Dabei können beim Ausschalten Spannungsspitzen entstehen. Der ESP hat extrem hochohmige Eingänge und fängt das ein.
Probiert mal einen Entstörkondensator parallel zum Relaisausgang oder einen Varistor. Alternativ wenn die Last es erlaubt, ein Solid State Relais. Das schaltet beim Nulldurchgang der Wechselspannung.

Noch ein paar Details mehr hier:
https://blog.thesen.eu/esp8266-reset-probleme-loesen-und-relais-stabil-schalten/

mi-hol, 2016-12-05:
@JanGoe könnte man den genannten Entstörkondensator in den Bausatz integrieren?

NemoN, 2016-12-05:
Ich werde die Tage es mal so bei meinem Board testen. Achtung ungetestet, Schaltung kann durchaus falsch sein ;-) Das Board und die beiden Verbraucher (Lampen) werden von einem gemeinsamen Kabel versorgt:

[image]

Schwarz die Verbindungen direkt auf der Platine...

famthiem, 2016-12-05:
Hallo 

habe im Moment wenig Zeit und verolge daher den Thread nicht sehr gut.

@stefan73,
bei mir machte der ESP sporadisch ein Reboot auch ohne, dass die Relais geschaltet wurden.
Seitdem ich die Reset-Leitung an VCC beschaltet habe,
    https://blog.thesen.eu/esp8266-reset-probleme-loesen-und-relais-stabil-schalten/
(blog: Problem 1) ist es mir nicht noch mal aufgefallen.
Ich muss aber dazu sagen, dass der ESP erst seit 2 Wochen so läuft.

Allerdings hatte ich heute einen TCP-Ausfall zu verzeichnen (ich mache alle 15 Minuten einen ping und logge das Resultat in einer Datei).
Ich musste einen Hard-Rest (VCC abgeklemmt) durchführen.

Beste Grüße

famthiem, 2016-12-06:
Hallo Dominik.

Ich arbeite mal die Nachrichten weiter ab...

Ich finde es garnicht schlecht, dass beim Starten die Relais anziehen. So erkennt man, dass der ESP unfreiwillig neu startet.

Angenommen das Licht ist an. Dann geht es aus, weil er resettet. Wenn das Licht danach wieder an ist, war das hochfahren erfolgreich. Wenn es aus bleibt, dann hängt er. Oder sehe ich das falsch?

VG

JanGoe, 2016-12-06:
Hallo,

Sehr intressant, wie du es gelöst hast den ESP Stabiler zu machen, ich werde mich das mal anschauen, und anschließend in V3 einarbeiten 

leider gibt es zurzeit noch ca. 500 pcb's der V2 deswegen wird eine V3 wohl noch etwas auf sich warten lassen

JanGoe, 2016-12-06:
@stefan73 eigentlich sollte es doch möglich sein die Varistoren auf dem board direkt gegen den X2/FX2 Kondensator zu tauschen an den ausgängen oder ?

NemoN, 2016-12-06:
> Angenommen das Licht ist an. Dann geht es aus, weil er resettet. Wenn das Licht danach wieder an ist, war das hochfahren erfolgreich. Wenn es aus bleibt, dann hängt er. Oder sehe ich das falsch?

Garagentor Steuerung: ESP stürzt ab und startet neu - Tor bekommt nachts um 3 Uhr einen Impuls und geht auf 8-)

NemoN, 2016-12-06:
@JanGoe @stefan73 

> eigentlich sollte es doch möglich sein die Varistoren auf dem board direkt gegen den X2/FX2 Kondensator zu tauschen an den ausgängen oder ?

Hatte ich auch schon überlegt. Das wäre das einfachste.

mnmn2, 2016-12-06:
@prikril 
I've checked the schematic, and it seems, that the basic idea was to use transistors to switch the relays, so there is only a "current-limiter" resistor between the GPIO and the transistor.
Hence, the board is using FETs to switch, so it's really needed to use pull-down resistors on the gate also.
I've made big-ugly :) cross-soldered resistors (wire leaded ones), and it's much better, there are no unwanted status-alternations on the relays.
(by the way, with the wire leaded resistors I was able to  "skip" the GPIO2 port problem, using an other port to switch the relay).
Based on the ESP documentation, I use 330Ohm in series to the gate, and 10kOhm between the gate and GND.

Thanks for the advice.

famthiem, 2016-12-06:
@NemoN 
ok, das Beispiel mit der Garagentorsteuerung ist überzeigend :-)
in der Garage habe ich aber Platz, da kommt ein RaspberryPi rein...

stefan73, 2016-12-06:
@JanGoe  @NemoN 
Wenn das Board genug Platz für die X2 Kondensatoren mit >=250V und 0,33 oder 0,47uF hat, sollte es gehen. Dazu bitte noch die Reset Leitung über ca. 3-10k auf 3.3V ziehen. Damit sollte erst mal das gröbste erledigt sein.
Ansonsten können lange Leitungen an GPIOs oder Reset in der Nähe der Relais ungünstig sein. Ich vermute aber, dass Ihr mit dem obigen Vorschlag schon einen deutlichen Unterschied seht.

Wegen der schaltenden Relais beim Start: Einfach andere GPIOs nehmen. Beispielsweise GPIO 5, 12 und 13 schalten beim Boot nicht.

Gruss Stefan

NemoN, 2016-12-15:
Mit den Tipps hier (GPIO Pullups, 100nF, 3300uF und X2 Kondensatoren) läuft mein TestBoard mit einer 220V LED Lampe und einer 220V Halogen-Lampe nun seit 100+ Stunden stabil. Vorher resetete das Board beim Schalten der Relais oder nach ein paar Minuten einfach so...

JanGoe, 2016-12-16:
kannst du mal ein bild von deine platine posten mit den "updates" ?

mi-hol, 2017-01-01:
@NemoN eib Bild waere toll :)

NemoN, 2017-01-19:
- Die X2 Kondensatoren sind auf dem Bild nicht zu sehen, die sind direkt an den LED Lampen  Anschlüssen.
- Der 3300uF Kondensator ist auf der anderen Seite

[image]

ronopitz, 2017-02-02:
Hallo,

ich möchte mit dem wifi relay einen Rollladen steuern. Benutze dazu die Platine mit zwei Relais.
In diesem Thread stehen sehr viele Informationen, leider kann ich nicht erkennen, welche sich auf Version 1 und Version 2 beziehen oder ob es eine Idee für eine zukünftige Version ist.

Ich benutze Version 2 und habe das Problem, dass beide Relais beim Booten anziehen und somit der Rollladenmotor Strom für hoch- und ruterfahren bekommt.

Dies hört sich wie eine Lösung an:
"Also bei mir ist es ohne Pull-Down Widerstand so, dass die Relais anziehen, wenn der ESP startet.
Das ist, weil die GPIOs standardmäßig auf Input stehen.
Mit dem Pull-Down des Transistors wird das Relais (NO) offen gehalten."

Was muss dafür verändert werden ?
Was kann/muss ich sonst noch machen?
Wie habt ihr dies realisiert?

Danke + schöne Grüße
Ron

prikril, 2017-02-02:
@ronopitz 
Hallo Ron,
also du hast dann wohl auch die NO (Normally Open) Version gekauft, die stromlos geöffnet ist.
Die GPIOs stehen beim Booten auf Input, wodurch ein Strom an den MOSFETs anliegt.
Wenn du dir mal den Schaltplan anschaust (https://github.com/JanGoe/esp8266-wifi-relay/blob/master/pics/schema.png?raw=true), siehst du, dass diese dort mit T1 und T2 markiert sind.
Du kannst die Steuerleitung (zwischen den MOSFETs und R3 bzw. R4) mit einem 120k oder 130k Ohm Widerstand gegen GND ziehen. (Anlöten an den SMD Widerstand auf der MOSFET Seite sollte am einfachsten gehen. GND ist an vielen Stellen möglich.)
Achte bitte darauf, dass die Beine des Widerstands keine anderen Stellen berühren. (am besten z. B. mit Schrumpfschlauch isolieren)
Leider klappt das aber nur beim Relais 2 gut.
Relais 1 kann weiter sehr kurz bzw. mehrmals sehr kurz anziehen, das dessen MOSFET mit GPIO 2 verbunden ist. (siehe https://github.com/JanGoe/esp8266-wifi-relay/issues/46#issuecomment-261736383)
Somit kannst du dir die Arbeit an Relais 1 eigentlich sparen und dich nur um das Relais 2 kümmern.
Wenn nur eine Seite kurz anzieht, sollte das aber keine Probleme machen oder?
Wenn doch müsstest du die Bahn von GPIO 2 zum R3 unterbrechen, einen anderen geeigneten GPIO aussuchen, mit R3 verbinden und das Programm anpassen.
Gib bitte Rückmeldung, ob es so funktioniert hat. Danke.
Sollte ich mich irren, darf sich jeder hier kurz melden. Dann passe ich meine Antwort umgehend an.

Gruß
Dominik

NemoN, 2017-02-15:
Mit den Änderungen auf meinem Board läuft es bereits seit 600+ Stunden (25 Tage) ohne Neustart...

kellermaaan, 2017-02-25:
Hi NemoN, könntest du vielleicht eine kleine Anleitung schreiben was du alles gemacht hast? Das würde mir zumindest echt helfen, weil ich das auch gerne mal ausprobieren möchte!!

Luettis, 2018-02-04:
[image]

Luettis, 2018-02-04:
Hallo, 
da ist es. Das Ergebnis meiner Lötarbeiten.
Nur leider verweigert  die Platine ihr Funktionsfähigkeit. WLAN funktioniert noch, sost nichts.
Ziel dieser Lötarbeiten war den ESP zu stabilisieren und das Anziehen der Relais beim Starten zu unterdrücken. Schwierigkeiten hatte ich eim Anlöten des linken Kondensators  am ESP(sieht man auch).
Alle anderen Lötstellen gingen schnell von der Hand. 
Als Widerstände hatte ich nur welche mit  100 KOhm anstatt der aus dem Text empfohlenen 120 bis 130 KOhm.  Änderung des Relais 1 an GPIO 5. Die Leitung GPIO 2 habe ich kurzerhand getrennt, nachdem der erste Testversuch (Anschluss an 220 Volt , relais sollten dann mit mit 3,3 volt bestromte LEDs zum Leuchten bringen, was im Vorfeld auch immer funktioniert hat) gescheitert war. 
2 Möglichkeiten:
Die Umsetzung der Empfehlungen falsch gemacht, falsche Lötpunkte etc. 
oder:
beim Anlöten des linken Kondensators am ESP zu lange gebraucht und den ESP zerstört. 

Schaut ihr euch die Lötarbeiten mal an und teilt mir mit wo ich es vergeigt habe. 
Vielen Dank für eure Hilfe

JanGoe, 2018-02-04:
Hallo,

irgendwie schaut es am ESP so aus das es einen "kurzen" gibt zwischen gnd und den "unteren" kontakten am ESP auf dem Bild das "unten" Links

Luettis, 2018-02-04:
Hi, 
das habe ich jetzt mehrfach mit der Lupe kontrolliert. Da wirds nicht dran liegen. Vermutlich dann doch zu heiß geworden.  Ansonsten aber  alles richtig gemacht?
Deswegen habe ich auch gleich zwei bestellt. ::-)))

Luettis, 2018-02-05:
Muss man eigentlich beim anlöten der Kodensatoren den Umweg über das ESP-Gehäuse machen?  oder könnte man alternativ den Reset PIN direkt an VCC anlöten. Müsste dann ein anderer Kondensator als der 104 genommen werden

Luettis, 2018-03-17:
[image]

Luettis, 2018-03-17:
[image]
[image]
[image]

Luettis, 2018-03-17:
Hallo!
Für alle, die wie ich eine Vorstellung haben aber bei der Ralisierung des Projekts bisher gescheitert sind. 
Ich benötigte eine verlässliche Rollladensteuerung, die per App/WLAN, Zeitsteuerung und manueller Tastersteuerung funktionieren  und zu allem Überfluss auch noch in die Schalterdose passen sollte.  
Zwangsläufig musste ich dann auf die von J.G. angebotene  Platine stossen. Ich habe mehrere Platinen bestellt bzw., bstellen müssen. Die erste Platine war bis auf das Netzteil fertig gelötet. Die anderen Platinen waren im Rohzustand,also nur Platinen und die anzulötenden Teile seperat.   Voller Tatendrang installierte ich meinen Sketch auf Platine 1 und schloss sie ohne weitere Tests direkt an den Rollladenmotor an. Fortan war ich einige Tage mit der Reparatur der Rolllade beschäftigt. Wie ich im Nachhinein lesen musste hatte die gleichzeitge Relaisschaltung die Endschltpunkte des Motors gelöscht und wegen der ungünstigen Position der Rollade mit der falsch angeschlossenen Richtung wickelte sie sich verkehrt herum auf die Welle. 
Nun denn, im Folgenden bin ich  hier im Forum gelandet und habe  von den bestehenden Problemen gelesen. (z.Bsp. das die 2 Relais bei Stromzuführung kurzfristig anziehen, s. Bsp. Garagentor)   Daraufhin habe ich die angebotenen Lösungen, so wie ich sie verstanden habe, versucht umzusetzen. Nur für jemanden aus einem  kaufmännischen Beruf war vieles sehr unverständlich und somit mit vielen Stunden des Lesens einhergehend. 
Um anderen diese Arbeit ein wenig zu vereinfachen  versuche ich hier in wenigen Worten meine Vorgehensweise, die zu einem erfolgreichen Abschluß geführt haben, zu beschreiben. 

Für das  Anlöten der 100 nF Stützkondensatoren an den ESP gibt es Bilder und war so einfach nachzuvollziehen.. Die Ausführung ist dafür vermutlich umso schwieriger. 
Die Beinchen an die Metallabschirmung anzulöten ist mir zwar gelungen.  Allerdings hat das so lange gedauert das es dem ESP definitiv zu heiss geworden ist.  Schnell war die Idee die Beinchem mit einem leitenden Metallkleber anzukleben geboren. Nur das Finden eines solchen Klebers in  einer empfehlenswerten Quallität und vernünftigen Menge ist mir nicht gelungen.  . Des Nachts ist mir die  Trennung  der Klebstoffe eingefallen. Mit einem Metallkleber habe ich die Beinchen der Stützkondensatoren festgeklebt. M.E. nach Trocknung  genau so fest wie löten. In einem weiteren Schritt für den Kontakt dann den Silberleitkleber aufgetragen.  (der klebt zwar, aber es fehlt an Festigkeit). Die  2. Beinchen habe ich dann ziemlich kurz abgeschnitten und einmal an Reset (untere Reihe ESP ganz links)  und das andere (ebenfalls die untere Reihe ganz rechts an 3,3 Volt) angelötet. Bei der anderen Platine, die ich komplett selber zusammengelötet habe, habe ich die Beinchen ich das Lötloch gesteckt und dann gelötet.
Problem 1 gelöst, Ergebnis  s. 1 Bild.
Das 2. Problem mit dem GPIO 2 ist eigentlich auch einfach. Entweder man trennt die Leiterbahn, die diagonal von der WLAN Antenne zu dem kleinen Widerstand führt. Das habe ich mit der Spitze des heißen Lötkolbens relaitv einfach hinbekommen. 
Oder bei der selbst zusammen gelöteten Platine den GPIO 2 Anschluss des ESP gar nicht erst  anlöten.. ebenfalls Bild 1, obere Reihe des ESP  3. Lötanschluss von rechts, deutlich zu sehen.  Dafür habe ich dann ein Kabel (Bild 1 das schwarze Kabel) an  GPIO 5 und an den Widerstand angelötet. 
Problem 2 abgehakt. 
Bis zu diesem Zeitpunkt funktionierte die Platine mit dem Sketch noch einwandfrei. Es ist sicher ratsam nach jeder Veränderung einen Test vorzunehmen. Dann weiß man zumindest wo der Fehler zu suchen ist.
Denn  Problem 3:
da die Platine, wie bereits erwähnt, für einen Rollladenmotor gedacht ist, ist das gleichzeitige Anziehen beider Relais für den Motor  nicht wirklich prickelnd. Es kann den Motor im schlimmsten Fall zerstören oder die Endpunktschaltung löschen. Bei mir war beides passiert; mit der Folge das ich mehrere Lamellen als auch den Motor austauschen musste.  Zur Verhinderung des Anziehens der Relais sollen 
 an die zwei 100 kOhm Widerstände auf der Mosfetseite  entweder 120 ode 130 kOhm Widerstände angelötet und mit GND verbunden werden.  Getan und nichts ging mehr.  Ich war mir auch nicht sicher ob ich den Hinweis wo die 120/130  kOhm Widerstände angelötet werden sollten  richtig verstanden hatte. Aber auch nach mehrmaligen Lesens bin ich immer zu dem selben Ergebnis gekommen.  Folglich die 100 kOhm Widerstände entlötet, durch neue  ersetzt und mit  dem den nächsten Wert (130 kOhm) an GND gelötet; ebenfalls erfolglos. Daraufhin
habe ich alle Teile entlötet und durch neue ersetzt.   Aber der Platine war nach anlöten der 120 bzw 130 kOhm Widerstände keine Reaktion mehr  zu entlocken.  
Verunsichert, ratlos und fast schon die Flinte ins Korn werfend suchte ich Hilfe bei einer hier auf eigenen Wunsch hin  nicht genannten Person, der aber umso mehr mein Dank gilt. Denn zum Einen bestätigte er mir die von mir vorgenommenen Lötarbeiten als richtig und gab mir den entscheidenden Hinweis die 100 kOhm Widerstände gegen 300 Ohm Widerstände auszutauschen und mit 10 KOhm an GND zu schliessen. Das steht hier zwar auch im Forum, aber das war für mich wegen den extremen Grössenunterschieden  so abwegig das ich es niemals in meine Überlegungen mit einbezogen habe. 
Aber die zwei verbliebenen Platinen funktionieren mit dieser Massnahme einwandfrei. 
[... cut ...]
Ein Nachmachen sollte allerdings gut überlegt sein. Bei mir ist der Harz in die Schraubklemmen geflossen und hat sie somit unbrauchbar gemacht. Es hat einiges an Arbeit gekostet die Schrauklemmen wieder nutzbar zu machen.  
Bild 3 zeigt noch den 3300 mF Elko und zusätzlich habe ich noch 2 Kühlkörper auf das Netzteil geklebt. 

Diese Massnahmen haben bei mir zum Erfolg geführt. Ob das aber eine Allgemeingültigkeit hat kann ich nicht sagen und ich  übernehme auch keine Garantie oder Haftung dafür.