JanGoe/esp8266-wifi-relay #50: Einstellungen über Browser?

JanGoe, 2016-07-14:
Hallo,

Es wäre wirklich schön wenn man sämtliche Einstellungen in der actuator.lua über den Browser in einem schönen Webinterface ändern könnte und dort ggf auch die Relays Schalten könnte so wie Temperaturen von angeschlossen Sensoren anschauen könnte 

da das eine etwas größere Sache ist evtl hat ja jemand lust hier was zu programmieren ? 
Würde mich dafür sicherlich auch bedanken in form von Wifi-Boards

JanGoe, 2016-07-14:
https://github.com/RedFoxy/ESP8266-Web-Interface

evtl könnte man hierrauf aufbauen ?

xCite1986, 2016-12-12:
hallo @JanGoe 

vielleicht wäre es besser das projekt und die anleitungen auf basis der espeasy-firmware aufzusetzen. Ich betreibe deine platinen mit der R95 und diese läuft wirklich stabil und die anbindung ist via mqtt ind interner API recht vielseitig und einfach. 

lg alex

JanGoe, 2016-12-12:
habe nun etwas mit der R120 und der R147 gespielt, leider gibt es bei diesen beiden version die rules nicht ? oder muss man diese als "plugin" irgendwie laden ?

xCite1986, 2016-12-12:
"rules" muss man erst unter  "Tools" > "Advanced" aktivieren

https://forum.pimatic.org/topic/1577/espeasy-add-a-relay/38