JanGoe/esp8266-wifi-relay #52: Where can I buy it?

snizzleorg, 2016-10-28:
Do you offer the hardware to buy somewhere?

NemoN, 2016-10-28:
check:

https://raspiprojekt.de/en/buy-it/shop/bausaetze/wifi-relais-v2-zweifach.html
https://raspiprojekt.de/en/buy-it/shop/restbestaende/wifi-relais-einfach.html

prikril, 2016-11-19:
You can also buy it directly from the inventor:
http://www.ebay.de/itm/ESP8266-ESP12-WiFi-WLan-Relay-IOT-Unterputz-Montage-moglich-/322315899088

Be carful:
There are two different versions of the V2 board.
The V2 is available with NC (normally closed) or NO (normally open) relays.
You should know which version you want.
Ask the inventor or the seller which version you could get.

JanGoe, 2016-11-19:
at the moment only the NO Version is available

prikril, 2016-11-19:
But raspiprojekt.de seems to sell the NC version.
Quote: "Die Relaiskontakte sind ohne anliegende Spannung geschlossen (NC)."

JanGoe, 2016-12-06:
at the moment you can buy it on ebay

http://www.ebay.de/itm/ESP8266-ESP12-WiFi-WLan-Relay-IOT-Unterputz-Montage-moglich-/322348058564?var=&hash=item4b0d7107c4:m:mdCNNovJ_sLy6FCYohtz6Qg

closed 2016-12-06 by JanGoe