JanGoe/esp8266-wifi-relay #53: File Update via tcp.php = problem/corrupt files

famthiem, 2016-11-19:
Hi.
When uploading a file (lua-script) with tcp.php, the file is not transferred completely. It seems that there are some sections not uploaded. 
I am using Windows 7. 

Thanks and best regards
Jörg

[Thiem_2016-11-12.zip](https://github.com/JanGoe/esp8266-wifi-relay/files/601336/Thiem_2016-11-12.zip)

prikril, 2016-11-19:
Hi,
at first I would recommend to backup your ESP.
Then you should copy the newest files from this repository to your ESP with TTL cable.
It seems that you are using an old version of the project.
Second thing:
I commented here https://github.com/JanGoe/esp8266-wifi-relay/issues/46#issuecomment-232938899 about the file transfer.
Do you speak German?
See my fork of the "ESP8266 WiFi File Manager" for more Details in English.
https://github.com/prikril/ESP8266_WiFi_File_Manager
Please let me know if that works for you.

Dominik

famthiem, 2016-11-19:
Hi Dominik.

Thanks for your response.
I have seen your File Manager. Neverteheless, I don't kow how to set up a php server..

I also tried to upload the newest scripts to the ESP via TTL cable (ESPlorer).
But I always got an error that there isn't enough RAM for this.
I own the first release of the wifi-relay. 

Thanks and regards
Jörg

PS: We may switch to german.

prikril, 2016-11-19:
Für Windows und Mac bietet sich XAMPP an.
Für Linux gibt es fertige Pakete die mit apache oder nginx kombiniert werden können.
Was hast du für ein System?

Wie hast du denn jetzt die tcp.php verwendet?
Da musstest du ja zumindet PHP für die Kommandozeile verwenden oder?

Und über das TTL Kabel sollte es auch mit der alten Version klappen.
Der ESP ist ja der gleiche, nur das Board darunter ist anders.

Wenn du im ESPlorer mal auf FS Info gehst und dann noch auf Reload, siehst du, wieviel Speicher aktuell frei ist und welche Dateien drauf liegen.
Durch Fehlübertragungen kann das Dateisystem des ESP durcheinander sein.
Da solltest du mal mit "Format" alles löschen. (Zur Sicherheit kannst du alle Dateien vorher auf deinen Rechner kopieren.)

JanGoe, 2016-11-19:
solltest du noch ein altes board haben ( Platine ist blau ) würde ich empfehlen die aktuelle firmware zu flashen die du hier -> https://github.com/JanGoe/esp8266-wifi-relay/blob/master/firmware/nodemcu-master-12-modules-2016-01-20-21-33-33-float.bin
findest und anschließend alle LUA Dateien
danach sollte es auch mit der tcp.php funktionieren