JanGoe/esp8266-wifi-relay #54: GPIO2

xboxpro1, 2016-12-06:
Hallo, laut Quellen im Internet muss GPIO2 HIGH sein beim booten. Laut Plan steuert GPIO2 über den FET aber Relais 1.  Ist der Plan oder sind die Quellen nicht korrekt?

JanGoe, 2016-12-06:
GPIO2 hat beim starten immer wieder "komische" zustände so das es hierzu zu problemen kommen kann, zur zeit wird gpio2 für das Relais1 genutzt 

für v3 ist geplant GPIO2 gegen einen anderen zu tauschen

xboxpro1, 2016-12-06:
OK, Danke.. Werde dann GPIO2 tauschen, wenn das Teil bei mir ist. Habe eines über eBay erstanden "xcancer777". Habe vor das hier zu verwenden..
<https://bitbucket.org/xoseperez/espurna>
Sollte mit Modifikationen der Ports funktionieren.

NemoN, 2016-12-06:
Bei der V2 habe ich bei einem Board GPIO 2 und 13 getauscht:

Gelb = Leiterbahn durchtrennen
Grün = Verbinden ;-)

[image]

xboxpro1, 2016-12-06:
Perfekt, danke. Ich würde noch GPIO2 auf HIGH legen.

xboxpro1, 2016-12-09:
Habe hier ein Schaltbild für die korrekte GPIO Beschaltung.

[image]

closed 2016-12-09 by xboxpro1