JanGoe/esp8266-wifi-relay #55: Works with ESPEasy very well!

mnmn2, 2016-12-06:
Just wanted to let you know, that the board works very good with ESPEasy (R120).

JanGoe, 2016-12-11:
can you post your settings, how the Switch can take the relay on/off ?

JanGoe, 2016-12-12:
okay, i have write i simpel php script for it

in config set the protocol to Generic HTTP
in tab Hardware set Pin12,13 (D6,D7 ) to input
in tab devices add a simpel Switch select GPIO12,13 Value-Name set to the GPIO witch control the relay ( 2,14 ) 

and use this php script:

`<?php
$was = $_GET[value];
$gpio = $_GET[valuename];

if($was != "" && $gpio != "")
{
echo "test okay $was / $gpio";
$file = file_get_contents("http://192.168.2.134/control?cmd=GPIO,$gpio,$was");

}
?>`

xCite1986, 2016-12-12:
hey @JanGoe 

i've done this with an espeasy-rule

```
on Switch_1#Switch do
  if [Switch_1#Switch]=1
    gpio,2,0
  else
    gpio,2,1
  endif
endon
```

and this work as well without an delay

more details about [here](https://forum.pimatic.org/topic/1577/espeasy-add-a-relay/39)

mnmn2, 2016-12-12:
@JanGoe : 
your sample is perfect, mine would be too complicated, because it drives a Hoermann garage door opener.