JanGoe/esp8266-wifi-relay #56: please add timer to Wifi hotspot

tkoeberl, 2016-12-15:
Hi,
after a power failure the access point is slower than the ESP's. The ESP's power up in hotspot mode. 
Can you please add  the functionality to reset the ESP after X minutes if it is in hotspot mode?

thanks,
-thomas

prikril, 2016-12-15:
Here is a quick and dirty hack (UNTESTED) for you, restarting the ESP after 3 minutes.
Add these lines at the end of "servernode.lua":
```
TMR_AUTORESTART_ID = 1
AUTORESTART_IN_MINUTES = 3
AUTORESTART_TIMER_ENABLED = true
tmr.register(TMR_AUTORESTART_ID, AUTORESTART_IN_MINUTES*60*1000, tmr.ALARM_SINGLE, function()
  if(AUTORESTART_TIMER_ENABLED == true) then
    node.restart()
  end
end)
```

tkoeberl, 2016-12-15:
cool, this should go into the sources.

prikril, 2016-12-15:
So it worked?

tkoeberl, 2016-12-15:
Sorry, dont have time to test it. end of year in IT business.

prikril, 2016-12-15:
ok, no problem.

tkoeberl, 2016-12-28:
Hi, its working, thanks.