JanGoe/esp8266-wifi-relay #57: Rolladensteuerung programmieren

HansWurst90, 2017-03-30:
Hi,
Ich möchte ähnlich wie in #35 die Platine zur Steuerung eines Rolladenmotors nutzen.
Wie genau läuft die Konfiguration dazu ab?
Muss ich die Logik dazu ( 1 MQTT Topic für beide Relais und das Verhindern des gleichzeitigen Einschaltens beider Relais ) selber programmieren oder gibt es dazu in der Weboberfläche Auswahlmöglichkeiten?

andreasotto, 2017-06-11:
+1
Genau das interessiert mich auch. Die beiden Relais müssten gegeneinander gesperrt werden können softwareseitig. Ich würde mich sehr über weitere Hinweise freuen.

prikril, 2017-06-12:
Hallo ihr zwei,
@HansWurst90
Nein über die Weboberfläche kann das nicht eingestellt werden.
Wenn du den Ordner https://github.com/JanGoe/esp8266-wifi-relay/tree/master/lua-mqtt anschaust, siehst du, dass da gar keine Oberfläche existiert.
Die gibt es nur im lua-tcp Teil und auch dort ist diese nur für die Erstkonfiguration vorgesehen.
Der MQTT Code vom aktor.lua ist schon seit Ende 2015 nicht mehr angefasst worden.
Ich glaube damit erübrigt sich auch deine Frage @andreasotto oder?
(Übrigens gibt man ein "+1" indem man auf das Symbol oben rechts eines Beitrags klickt und den Daumen hoch auswählt.)
Man müsste ähnlich wie ich letztes Jahr beim lua-tcp Teil (siehe https://github.com/JanGoe/esp8266-wifi-relay/commit/1c6706af3426f7cb2c1a6ee3a3ae23956d1b14d0) den MQTT Code mal komplett überarbeiten.
Aktuell habe ich aber keine Zeit und mache auch nichts mit MQTT.
Ihr müsstet also selbst ran.
Wenn ihr Lua könnt, sucht doch mal im actuator.lua (https://github.com/JanGoe/esp8266-wifi-relay/blob/1c6706af3426f7cb2c1a6ee3a3ae23956d1b14d0/lua-tcp/actuator.lua) nach "INTERLOCK".
Beachtet bitte auch, dass die Verwendung der Timer seitens NodeMCU in Zunkunft geändert werden wird. (siehe https://nodemcu.readthedocs.io/en/master/en/modules/tmr/)
Falls ihr die alten Nummern (0-6) weiter verwenden möchtet, müsste ihr bei Updates der Firmware darauf achten, dass diese so noch verwendet werden können.
Ansonsten ist der objektorientierte Ansatz mit `tmr.create()` zu verwenden.
Dazu braucht euer ESP aber eine aktuelle Version von NodeMCU, da früher (genaues Datum siehe Commits von NodeMCU) die Funktion `tmr.create()` noch nicht existiert hat.

Als Codeansatz folgender Vorschlag:
Damit nicht das passiert, was offensichtlich früher mal der Fall war, nämlich dass beide Relais gleichzeitig an waren, müssen beim Wechsel beide erst ausgeschaltet werden.
Pseudocode mit Annahme, dass Relais1 das Rollo hoch fährt und Relais2 runter:
```
-- static vars
DIRECTION_STOP = 0;
DIRECTION_UP = 100;
DIRECTION_DOWN = 200;
-- function definitions
-- Hier fehlen noch die Funktionen für das An- und Ausschalten der Relais!
function drive(direction)
  -- first switch off both relays to prevent damage of the roller blind
  relay1.switchOff();
  relay2.switchOff();
  if (direction==DIRECTION_UP) then relais1.switchOn();
  if (direction==DIRECTION_DOWN) then relais2.switchOn();
end
-- program start
mqtt.listen(Feld für Rollo als Variable data einlesen) {
  if (data=="STOP") then drive(DIRECTION_STOP);
  if (data=="UP") then drive(DIRECTION_UP);
  if (data=="DOWN") then drive(DIRECTION_DOWN);
}
```
Wie ihr seht, spart ihr euch das "Interlock", weil es hier rein um die Rollosteuerung geht und daher besser immer zuerst beide Relais ausgeschaltet werden, wenn eine Fahranweisung kommt.
Die Frage ist jetzt noch, wie lange die Relais jeweils angezogen bleiben sollen.
Irgendwann ist das Rollo ganz auf oder zu.
Es bäuchte auch einen Timeout, falls z. B. die WLAN Verbindung verloren geht.
Spätestens hier würden dann die Timer zum Einsatz kommen.
Im zuvor genannten actuator.lua gibt es sowas und hab ich "delay_timers" genannt.
Allerdings wird da mit globalen Variablen gearbeitet, was man besser nicht macht.
Auch fehlt noch die Möglichkeit das Rollo per Taster/Schalter zu steuern und nicht nur per MQTT.
Die Rückmeldung per MQTT, wenn ein Relais anzieht bzw. wenn eine Fahrt beendet ist, fehlt auch noch.

Eine Universallösung, die alle Wünsche abdeckt, wird es eh nicht geben.
Das ist jeder selbst angehalten den Code entsprechend noch anzupassen.
Könnt ihr beide programmieren?
Vielleicht kann der lua-tcp Code so umgebaut werden, dass er mit MQTT funktioniert.
Dazu müsstet ihr euch in die Doku von NodeMCU einlesen.

Noch ein letzter Punkt:
Bitte beachtet, dass es von den Board anfangs eine NC Version und später dann eine NO Version gab.
Dementsprechend sollte der Code konfigurierbar sein.
Beim actuator.lua muss man aktuell nämlich gpio.HIGH und LOW vertauschen, falls man die NO Version hat.
Schöner wäre es, wenn ganz oben bei den "user defined options" eine Variable die Version, also ob es ein Board mit NC oder NO Relais ist, angibt und der restliche Code darauf reagiert.

Waren das genug Hinweise?

brezel1965, 2017-09-17:
Hallo prikril,
puh, da haste einige Hinweise gemacht. ;-) Ich würde auch gerne das esp8266-wifi-relay für die Rolladensteuerung verwenden, meine rudimentären Programmierkenntnisse reichen vermutlich nicht aus um deine ganzen Hinweise umzusetzen. Für meine NO Version habe ich das actuator.lua aus dem lua-tcp Teil angepasst.
Das INTERLOCK macht aber bei mir nicht ganz das was es soll, es wird zwar erfolgreich verhindert das Relay2 schaltet wenn Relay1 geschalter ist, aber wenn Relay2 geschaltet ist, wird das Relay1 trotzdem geschaltet und Relay2 fällt sofort ab, danach sind die States durcheinander und Relay2 läst sich nicht mehr schalten. INTERLOCK und TIMER zusammen geht leider auch nicht. Werde mal eine neuere Firmware versuchen, vielleicht liegt es daran?

prikril, 2017-09-17:
Also an der Firmware sollte es nicht liegen, außer du verwendest Befehle oder Funktionen, die mit einer neuen Version eingeführt wurden.
Bei der neuen NodeMCU Version gibt es auch größere Änderungen.
Die Timer können zwar aktuell noch so verwendet werden wie früher.
Dieser Ansatz könnte zukünftig aber wegfallen.
Wenn du etwas programmiert hast, solltest du mal ein Repo für dein Projekt auf deinem Github aufmachen.
Clone einfach dieses Projekt hier und verändere dann den Code bei dir lokal.
Dann kannst du deinen aktuellen Code hochladen und andere Leute können ihn ansehen.
Praktisch ist auch, dass Github die Änderungen gleich farbig markiert so wie es Tools für GIT auch auf deinem Computer können.
So blind zu raten, was du gemacht hast und wo der Fehler sein kann, ist unmöglich bzw. extrem mühsam und würde dieses Issue zuspammen.
Da wäre es dann besser, wenn du in deinem Repo selbst ein Issue dazu erstellst, damit dir geholfen werden kann.
Sollte die aktuelle Implementierung dieses Projekt hier falsch oder fehlerhaft sein, kann ich das gerne mal anschauen, wenn ich Zeit finde.