JanGoe/esp8266-wifi-relay #60: Eagle PCB File

danielpondruff, 2017-08-29:
Hallo Jan,

ist es eventuell möglich dass du das Projekt vollständig freigibst, sodass wir die Platine selbst ätzen lassen können ?

Mit freundlichen Grüßen
Daniel Pondruff

NemoN, 2017-08-29:
@danielpondruff ich hätte noch ein paar V2 Module (NO Version) abzugeben...

JanGoe, 2017-08-29:
ich hätte noch mehr wie genug "leere" pcbs hier, in der V2 NO, die ich
günstig abgeben kann

foxel, 2017-08-29:
I'd also want to have PCB files.

Or some PCB's at least

akushneruk, 2018-10-09:
Hi! Can you send me pcb file and specification? Very interesting project. I want insert this for my smart home project.
kushneruk.andriy@gmail.com
Thanks!

JanGoe, 2018-10-09:
i can sell you the pcb's

best,
jan

burak76, 2018-10-24:
Hi jan,
Can you send me pcb file i want to try this project.My mail sburak@qq.com

mastadook, 2019-03-12:
Hi, 
hast Du noch fertige Module V2 oder aktueller?
Ich hätte gerne noch 2-3 Stück, machen sich gut und tun was sie sollen :)

Gruß
Clemens