JanGoe/esp8266-wifi-relay #61: Schalten via tcp.php funktioniert nicht. PHP5 geht nicht zu installieren

mastadook, 2019-03-08:
Hallo,

ich habe das Problem, das ich die Relais jetzt über ein Jahr im Schrank hatte.
SHC hatte ich abgebaut und durch Domoticz ersetzt.
Jetzt wollte ich mit den mal wieder straten was zu schalten.
Leider ist eine Integration in Domoticz wohl nur über tcp.php  und die Befehle möglich:
z.B. php tcp.php 192.168.xxx.xxx 2x4x1
Wenn ich das nun ausführe, dann wird in der Shell einfach nur die ganze tcp.php Datei abgespult und angezeigt.
PHP5 kann ich weder auf dem Raspian Stretch noch auf meiner Ubuntu Kiste installieren, wegen neuer Pakete.
Aber es ist ja PHP7 installiert.
Muss ich PHP5 drauf bekommen oder sollte das mit den normalen PHP7 Paketen auch klappen und wo kann noch mein Fehler liegen? 

Wäre um jeden Tipp dankbar

Schöne Grüße
Clemens

NemoN, 2019-03-08:
Was kommt als Ausgabe wenn du auf der Linux Console "php -v" ausführst?

mastadook, 2019-03-08:
PHP 7.0.33-0+deb9u2 (cli) (built: Feb 25 2019 23:13:19) ( NTS )
Copyright (c) 1997-2017 The PHP Group
Zend Engine v3.0.0, Copyright (c) 1998-2017 Zend Technologies
    with Zend OPcache v7.0.33-0+deb9u2, Copyright (c) 1999-2017, by Zend Technologies

Aber ich war so darauf fixiert, das es was mit der php version zu tun haben muss,
das ich gar nicht gesehen habe, das ich das ganz falsche File als tcp.php verwendet habe.
Habe dann mit eben mit 
wget https://raw.githubusercontent.com/JanGoe/esp8266-wifi-relay/master/tcp.php
die Datei nochmal an die richtige Stelle geladen und dann die richtige tcp.php genommen, und schon geht alles wie gewünscht.
Problem saß wie fast immer vor dem Bildschirm und an der Tastatur :)
Trotzdem danke das Du mir helfen wolltest.

Gruß
Clemens

closed 2019-03-08 by mastadook